Plippo/asus-wmi-screenpad #2: Error at step 4: headers not found

bphd, 2020-06-06:
```sh
$ sudo dkms build -m asus-wmi -v 1.0
Error! echo
Your kernel headers for kernel $KERNEL cannot be found at
/usr/lib/modules/$KERNEL/build or /usr/lib/modules/$KERNEL/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
$ ls /usr/lib/modules/$KERNEL/source
ls: cannot access '/usr/lib/modules/$KERNEL/source': No such file or directory
$ du -h /usr/lib/modules/$KERNEL/build/*
58M     /usr/lib/modules/$KERNEL/build/vmlinux
```

lakiluki1, 2020-06-06:
Make sure you have appropriate kernel headers installed, which usually come in a separate package. For example, on Arch it's installed with the `linux-headers` package.

bphd, 2020-06-06:
Theres no `linux-headers` but I installed `linux-lts-headers` because I run on an LTS kernel. After rebooting the command said that the module was already builded and installed (mysterious, considering that even the adding operation behaved the same way). But I don't have anything new under `leds/` so I have nothing to control the backlight for now

lakiluki1, 2020-06-06:
You probably need to remove the kernel module again and repeat the installation procedure, as per the "Removing or reinstalling" README section.
(Check https://github.com/Plippo/asus-wmi-screenpad#removing-or-reinstalling)

Once the kernel module is loaded and started you should also probably see some logs in the `dmesg` output, so I suggest you check the output of `dmesg | grep asus_wmi`.

You can also check if there's anything in `/sys/devices/platform/asus-nb-wmi`.

bphd, 2020-06-06:
```sh
$ ls /sys/devices/platform/asus-nb-wmi/
cpufv  driver  driver_override  hwmon  input  leds  modalias  power  rfkill  subsystem  uevent
```

but

```sh
$ ls /sys/class/leds/
asus::kbd_backlight  input3::capslock  input3::numlock  input3::scrolllock  phy0-led
```

and

```sh
$ sudo dmesg | grep asus_wmi
[    9.000668] asus_wmi: ASUS WMI generic driver loaded
[    9.004545] asus_wmi: Initialization: 0x1
[    9.004687] asus_wmi: BIOS WMI version: 9.0
[    9.004782] asus_wmi: SFUN value: 0x21
```

Plippo, 2020-06-11:
You have to install the correct headers for your kernel. Most distributions provide a meta package for that. If you reinstalled without errors but don't see the device, than the device probably isn't recognized. What model of the ZenBook Duo do you have?

bphd, 2020-06-11:
Not a ZenBook Duo

closed 2020-06-11 by bphd