Plippo/asus-wmi-screenpad #3: Support for other Screenpad devices

lakiluki1, 2020-06-06:
Hey, love the work you're doing here, it would be pretty cool to see full Screenpad support on Linux, since ASUS really can't seem to bother. 

I was wondering, since you now probably have the most insight in the inner workings of the screenpad wizardry - how difficult would it be to port this screenpad backlight brighness support over to other Zenbook models, with different versions of the screenpad?

For example, I'm running a `Zenbook UX434F` with screenpad 2.0. The only thing that really works is turning the screen on and off (it's recognized as an HDMI display device in Xorg, with basic power managment working). However, the resolution. refresh rate or brightness can't be changed.

So, I was wondering, if you have the time to do so, could you consider adding support for those additional devices? I can provide all the info you need for this specific model and contribute in any way I can, and I'm assuming others could too.

Thanks!

Plippo, 2020-06-11:
Hi, I can try to look into it, but it seems that it might be more complicated with your model (with the ScreenPad Plus, resolution and refresh rate could already be controlled out of the box, only brightness wasn't working).
But if you provide me with your DSDT, I can try if I see anything in there. Unfortunately I don't have much time at the moment to do much research, but maybe I see something if I compare it to the one from my device. Please send it to me by mail (mail AT philmerk DOT de) instead of posting it on github to prevent copyright problems.

Plippo, 2020-06-11:
I forgot to mention how to do it :) To export the DSDT, call:
```
cat /sys/firmware/acpi/tables/DSDT | sudo tee dsdt.dat
```

sbespalov, 2020-08-05:
:+1:  for other devices, would be nice to have support for `ASUS ZenBook Flip 14 UX463FL` in my case 

@lakiluki1 it may be out of topic but does it works just as touchpad with disabled screenpad display?

Plippo, 2020-08-12:
I received one DSDT for the UX480F some time ago, but it was very different from the one on the ZenBook Duo, so I only could give some hints to the sender for things to try out. I never heard back from him, so I don't know if it worked.

sbespalov, 2020-08-14:
@Plippo thanks for response
I could provide my `DSDT` for `UX463F` if you like.

Plippo, 2020-08-14:
Sure, just send it via email to the address I mentioned above and I'll look into it.

lakiluki1, 2020-09-06:
@sbespalov I'm a bit late to the party and have totally forgot about this - but yes, it works perfectly, and that's how I use it when on the go, as it's really battery hungry without adequate power management and the ability to turn the brightness or the resolution down. 

@Plippo I have sent you the relevant DSDT file for the Zenbook UX434F, thanks!

nichisoft, 2021-02-12:
> I received one DSDT for the UX480F some time ago, but it was very different from the one on the ZenBook Duo, so I only could give some hints to the sender for things to try out. I never heard back from him, so I don't know if it worked.

Hi Plippo,
I have an UX480F. In attachment you can see my dsdt.dat compressed in zip file.
My OS is Ubuntu 20.04.2 LTS with 5.8.0-43-generic kernel.

Thanks you so much.
Great job.

[dsdt.zip](https://github.com/Plippo/asus-wmi-screenpad/files/5972189/dsdt.zip)

Plippo, 2021-02-13:
Thanks for the DSDT, I've answered you by email.

Qonfused, 2023-03-30:
Based on the UX480F DSDT provided by nichisoft, there still exist the same WMI methods; the ASUS driver in windows still expects the same inputs + ranges.

Notably the backlight adust WMI method (`0x00050032`) is the same; it takes the same input (`IIA1`) between `0x00` and `0xFF` for setting the new brightness level.

Concerning the connector power WMI method (`0x00050031`), there are missing checks for connector power (read from the first bit of `REBC (0x12, 0x02)`, [ref](https://github.com/Qonfused/ASUS-ZenBook-Duo-14-UX481-Hackintosh/blob/main/docs/resources/screenpad-plus.md#wmnb-methods---dsdt)), which is used to read and restore the last backlight value (lines 55533-55543). Passing in the connector power (`IIA1`) determines whether to overwrite with `0xFF` and shut off connector power, or restore the previous backlight value.
^ The method used in the UX480F always overwrites this backlight value, so I'd recommend explicitly setting backlight values after toggling connector power for better compatibility. You could also execute and check the output of a `0x0050035` WMI method, which if present would write the last written value to the EC register.

If you implement an ACPI patch that addresses EC registers directly, note that the EC registers for the screenpad backlight are offset by 3, so the read address is shifted from `0x12` to `0x0F`, and the write address from `0x13` to `0x10`. Would need to confirm whether this holds for other screenpad devices, but this is more or less what I'd expect for the HDMI/trackpad-based screenpads.

I also wouldn't worry about copyright from sharing DSDTs as this falls under fair use and enables cooperative operability.

Plippo, 2023-04-11:
Thank you very much for these insights. You seem to have a much deeper understanding of ACPI and WMI than I do, for me it was a combination of trial and error with looking up specs back when I got my ZenBook (and tbh I have forgotten much of it). For me, using WMI methods seemed safer than directly writing to registers (it might not work for all models, but at least seems less likely to break anything).

If I remember correctly, after some experiments using acpi_call with @nichisoft, we unfortunately didn't manage to turn the power on (the brightness values could be set and the new values were returned by `0x00050032`, but the screen didn't turn on if we tried to call `0x00050031`). In the end I didn't hear back from them so I don't know if they found a way to make it work.

nichisoft, 2023-04-12:
Unfortunately, after all the various attempts, I have never been able to get my screenpad to work on the asus UX480F on Ubuntu 20.04 and 22.04. If you have any other suggestions I can try and give you feedback.