Plippo/asus-wmi-screenpad #8: allow for brightness=0 without switching off the screen

s-light, 2020-06-16:
currently you the screen to off if the brightness is set to 0.
as fare as i have seen this is handled in [asus-wmi.c#L700](https://github.com/Plippo/asus-wmi-screenpad/blob/master/asus-wmi.c#L700)
for me personal i would like to be able to set the brightness to 0 but leave the screen on -
if it is switched off my desktop moves the windows from the second screen to the first..
that is expected if i switch the screen of - but if i just want to save some power while not using it temporary i prefer no window movement ;-)

do you have any idea how to integrate this in a nice way?
is there a other / better possibility to toggle the on/off separate to the brightness?

Plippo, 2020-06-28:
The problem is that the leds interface only allows setting one value and doesn't support controlling screen on/off separately from brightness. As a workaround, you can try setting the brightness to 1 instead of 0. For me, this also completely turns off the backlight, without turning off the screen. Maybe this also works for you. When using the "real" brightness interface of the kernel (once the userspace issuses are resolved), we will be able to control the brightness separately from the on/off state.

Plippo, 2020-06-28:
I've just seen your Pull Request and realized that you also went for setting the value to 1 :) Looks good, merged it.

closed 2020-06-28 by s-light