Plippo/asus-wmi-screenpad #9: Doesn't seem to work on Arch

laurinneff, 2020-06-28:
I'm on Linux 5.7.6-arch1-1. I've done almost nothing after installing (just installed GNOME and asus-wmi-screenpad). My laptop's model number is UX481FLY. If you need more info, just ask

CampGareth, 2020-07-29:
Same boat here on Manjaro (Arch based) so I installed acpi_call for my kernel version and then used the raw ACPI calls from this thread: https://github.com/s-light/ASUS-ZenBook-Pro-Duo-UX581GV/issues/1

Now there's a couple of bash scripts on my machine like this:
```
#!/bin/bash

echo '\_SB.ATKD.WMNB 0x0 0x53564544 b32000500FF000000' | sudo tee /proc/acpi/call
```

Not quite as convenient but it'll do for now.

Plippo, 2020-07-29:
@laurinneff I'm really sorry I didn't answer earlier, I didn't log into Github for some weeks and somehow missed the email notification about the creation of this bug... Did you (or @CampGareth) get any error messages while trying to install the module or did it just silently fail?

CampGareth, 2020-07-29:
No obvious errors on build, I'll include my logs here:
[debugging-asus-wmi-screenpad.txt](https://github.com/Plippo/asus-wmi-screenpad/files/4997567/debugging-asus-wmi-screenpad.txt)
Running lsmod | grep "asus" pre and post install and reboot gives exactly the same results:
```
$ lsmod | grep "asus"
asus_nb_wmi            28672  0
asus_wmi               40960  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
rfkill                 28672  10 asus_wmi,bluetooth,cfg80211
battery                20480  1 asus_wmi
wmi                    36864  3 asus_wmi,wmi_bmof,mxm_wmi
i8042                  32768  1 asus_nb_wmi
```

I'm guessing it's a naming clash? Two asus-wmi modules?

Plippo, 2020-08-12:
I don't see anything suspicious in the logs either. Maybe the original asus_* modules are already compiled into your kernel and thus can't be replaced by the modified versions? It's normal that both asus_wmi and asus_nb_wmi are active, those control different parts of the device.
Are you able to map the keys above the touchpad, to the left of the on/off key? Those are also provided by the modified modules.

TRPB, 2021-07-02:
I  can report the same thing, it appears in /sys/class/leds but the brightness is reported as 0 even though the screen is on (50% I'd guess from looking at it) and changing the brightness has no effect.

Thank you for your work anyway, it led me here and the acpi_call method above which works fine. I wish I'd discovered it before I spent an hour looking through xrandr, ddcutil and perusing /sys/class/backlight.