Plippo/asus-wmi-screenpad #10: Doesn't seem to work on Linux Mint 20

Totone, 2020-08-11:
Hi
I bought the Duo (model: UX481F)& I installed Linux Mint 20.
I found this tool & tried to install it, but after reboot, anything new in `/sys/class/leds`
I tried to remove & reinstall the tool, anything changed.

```bash
work@duo:/usr/src/asus-wmi-1.0$ uname -r
5.4.0-42-generic
```
On last install I launched the script `prepare-for-current-kernel.sh` even if I have the right kernel, but result was the same.

Here are some console I/O during install which maybe can help:

```bash
work@duo:/usr/src/asus-wmi-1.0$ ls /sys/class/leds/
asus::kbd_backlight  input25::numlock     input3::capslock  input3::scrolllock
input25::capslock    input25::scrolllock  input3::numlock   phy0-led

work@duo:/usr/src/asus-wmi-1.0$ sudo dkms add -m asus-wmi -v 1.0

Creating symlink /var/lib/dkms/asus-wmi/1.0/source ->
                 /usr/src/asus-wmi-1.0

DKMS: add completed.
work@duo:/usr/src/asus-wmi-1.0$ sudo dkms build -m asus-wmi -v 1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.4.0-42-generic -C /lib/modules/5.4.0-42-generic/build M=/var/lib/dkms/asus-wmi/1.0/build...
Signing module:
 - /var/lib/dkms/asus-wmi/1.0/5.4.0-42-generic/x86_64/module/asus-nb-wmi.ko
 - /var/lib/dkms/asus-wmi/1.0/5.4.0-42-generic/x86_64/module/asus-wmi.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.
work@duo:/usr/src/asus-wmi-1.0$ sudo dkms install -m asus-wmi -v 1.0

asus-wmi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-42-generic/extra/

asus-nb-wmi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-42-generic/extra/

depmod...

DKMS: install completed.

```

Tendrid, 2020-08-11:
I'm running into a related problem.  I use Ubuntu 20, with cinnamon.  While im in Unity, this driver seems to work fine, but in cinnamon, the display is registered, but does not allow me to change the resolution (and the backlight doesnt seem to be working)

It should be really easy to debug with ubuntu 20 by following these steps: https://linuxconfig.org/ubuntu-20-04-cinnamon-desktop-installation

Please let me know how i can help in any way!

Totone, 2020-08-11:
My Mint is also with Cinnamon. Maybe it is the issue.
I'll try to install Mate later, maybe it solves the problem.
(btw, it's akward to don't understand why there is an issue)

Plippo, 2020-08-12:
When the screenpad is turned off using the modified kernel module, it's not visible in the display settings, this may have been the problem. As soon as you increase the brightness, it gets turned on and thus visible.

Totone, 2020-08-13:
But the screenpad isn't turned off. The behaviour is like anything had been done (

I didn't install MATE yet. I have to learn to manage my time.
I come back here after to give a feedback

Totone, 2020-08-20:
Well, it still doesn't work.

I removed Cinnamon (I didn't like it anyway) & reinstall Mint with MATE env.
I installed the module but, like before, there is no new input after reboot.

Well ok, I installed xfce desktop environment to maybe have a similar luck as Tendrid but nope.
So, why not, I installed Unity desktop env that time & tried to reinstall the module. It did not do, the behaviour is the same I decribed days ago.

I let the log for last try. I'm sure I'm doing something wrong, but I don't understand what.

[terminalOutput.txt](https://github.com/Plippo/asus-wmi-screenpad/files/5104637/terminalOutput.txt)

kevinchappell, 2021-06-14:
@Totone Seems this can be closed now?

Totone, 2021-06-15:
Yes sorry, problem solved (months ago)

closed 2021-06-15 by Totone