Plippo/asus-wmi-screenpad #11: Doesn't seem to work on Ubuntu 20.04

ualich, 2020-08-26:
I use ASUS Zenbook Duo UX481FL-BM067R with Ubuntu 20.04 and Linux kernel version `5.4.0-42-generic`.

I went through the installation process, but a problem similar to #10 emerges after installing the module to kernel. The directory `/sys/class/leds/asus::screenpad` is not generated on boot, but the keyboard leds cease to work - the reason being that the directory `/sys/class/leds/asus::kbd_backlight` also fails to be generated on boot.

I recovered keyboard led function by removing  the module  as stated  in `README.md`:
```
sudo dkms remove -m asus-wmi -v 1.0 --all
sudo rm -r /usr/src/asus-wmi-1.0
```

Totone, 2020-09-01:
Sorry for the long time without answer, I was on holiday, far far from my Zenbook.
So, the best I can suggest is to reinstall the module with some reboots between steps as I did.
I don't really know how in my case it worked to be honest, but maybe it'll be the same for you

ualich, 2020-10-11:
I have now figured that most likely the problem was in generating the machine owner key (MOK) during reboot. As the key was not properly configured the module did not build correctly on startup.

Closing this issue now.

closed 2020-10-11 by ualich

ilivss, 2021-07-25:
> I have now figured that most likely the problem was in generating the machine owner key (MOK) during reboot. As the key was not properly configured the module did not build correctly on startup.
> 
> Closing this issue now.

How do we generate a MOK properly then? How to fix this issue?

Plippo, 2021-07-26:
@ilivss  Maybe this solution helps? https://github.com/Plippo/asus-wmi-screenpad/issues/27#issuecomment-886238029