Plippo/asus-wmi-screenpad #13: Patch can't be applied to kernel 5.9.

axegon, 2020-11-04:
Running openSUSE tumbleweed.

This function in particular in [asus-wmi.c](https://github.com/Plippo/asus-wmi-screenpad/blob/master/asus-wmi.c#L737):

```C
static void asus_wmi_led_exit(struct asus_wmi *asus)
{
	if (!IS_ERR_OR_NULL(asus->kbd_led.dev))
		led_classdev_unregister(&asus->kbd_led);
	if (!IS_ERR_OR_NULL(asus->tpd_led.dev))
		led_classdev_unregister(&asus->tpd_led);
	if (!IS_ERR_OR_NULL(asus->wlan_led.dev))
		led_classdev_unregister(&asus->wlan_led);
	if (!IS_ERR_OR_NULL(asus->lightbar_led.dev))
		led_classdev_unregister(&asus->lightbar_led);
	if (!IS_ERR_OR_NULL(asus->screenpad_led.dev))
		led_classdev_unregister(&asus->screenpad_led);
	if (asus->led_workqueue)
		destroy_workqueue(asus->led_workqueue);
}
```

This fixes it and works like a charm:

```C
static void asus_wmi_led_exit(struct asus_wmi *asus)
{
        led_classdev_unregister(&asus->kbd_led);
        led_classdev_unregister(&asus->tpd_led);
        led_classdev_unregister(&asus->wlan_led);
        led_classdev_unregister(&asus->lightbar_led);
        led_classdev_unregister(&asus->screenpad_led);

        if (asus->led_workqueue)
                destroy_workqueue(asus->led_workqueue);
}
```

I could prepare a patch file for 5.9 in particular and tweak the `prepare-for-current-kernel.sh` script and make a PR. Or perhaps make branches for different kernel versions for such scenarios? @Plippo you're the maintainer, your call.

Plippo, 2020-11-05:
I would prefer adding a tweak to the script that checks the kernel version and applies the appropriate patch. This way you can just run the script without having to worry which branch to choose. It would be awesome if you created the PR. Thanks!

axegon, 2020-11-05:
Sure. Apparently it's the same story with 5.8, will make a patch for both and I'll send a PR in 2 minutes.

Plippo, 2020-11-06:
Thanks, pull request looks good.

closed 2020-11-06 by Plippo