Plippo/asus-wmi-screenpad #15: How to start proceed with asus-wmi-screenpad where the documentation stops?

hreuver0183, 2021-01-02:
I have a Asus S433F with screenpad.

I tried using the screenpad with Debian, sofar without succes.
My issue is that after following the commands and compiling the module, there is no /sys/class/led/asus::screenpad

Last attempt was with 5.10.x kernel (to keep things simple, unsigned for now):
`[    2.747929] asus_wmi: loading out-of-tree module taints kernel.`
`[    2.748848] asus_wmi: ASUS WMI generic driver loaded`
`[    2.749694] asus_wmi: Initialization: 0x1`
`[    2.749734] asus_wmi: BIOS WMI version: 9.0`
`[    2.749759] asus_wmi: SFUN value: 0x21`
`[    2.749761] asus-nb-wmi asus-nb-wmi: Detected ATK, not ASUSWMI, use DSTS`
`[    2.749762] asus-nb-wmi asus-nb-wmi: Detected ATK, enable event queue`
`[    2.750929] input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input11`
`[    2.809599] battery: new extension: ASUS Battery Extension`

Unfortunately searching on DSTS yields a lot of information, but no answers.
Any hints on how to proceed?

Plippo, 2021-01-03:
Well it might be that the S433F uses a different mechanism for the screenpad, or maybe just a slightly different format. If you want, you can do a dump of the DSDT and send it to me, then I can try to look into it (but I can't guarantee anything, it usually involves some guessing and experimenting). You can find information on how to dump the DSDT in this issue: https://github.com/Plippo/asus-wmi-screenpad/issues/3

hreuver0183, 2021-01-03:
Providing the using the DSDT is easy.
Reading the table is different, I could not make anyhting usefull out of it myself. 
` cat /sys/firmware/acpi/tables/DSDT > dsdt-dump.txt`
I hope you can do anything usefull with it.

Just for my information:
After the commands in the README what should you see in the dmesg if everything works correctly?

Thanks for your effort in advance.

[dsdt-dump.txt](https://github.com/Plippo/asus-wmi-screenpad/files/5762164/dsdt-dump.txt)

Plippo, 2021-01-06:
Thanks for the DSDT. Unfortunately, the WMI methods that are used on my machine to control the brightness are not available in your WMI. I've looked through your WMI and haven't found any methods that look like they could be candidates, I'm sorry.

About your question: There's nothing you can see in the dmesg, mine looks exactly like yours.

hreuver0183, 2021-01-06:
That is very unfortunate for me.
But thank you for your effort, I really appreciate your help.

closed 2021-01-06 by hreuver0183