Plippo/asus-wmi-screenpad #16: Issues with Kernel 5.10.10

da-moon, 2021-01-26:
I followed the steps as it was explained in the README file. My current Kernel version is `5.10.10-051010-generic` and I am running pop_os 20.10 distro. I got the following error

```
patching file asus-nb-wmi.c
Hunk #1 succeeded at 578 with fuzz 1 (offset 109 lines).
Hunk #2 succeeded at 588 (offset 111 lines).
Hunk #3 succeeded at 602 (offset 111 lines).
patching file asus-wmi.c
Hunk #1 succeeded at 32 with fuzz 2.
Hunk #2 succeeded at 187 (offset 8 lines).
Hunk #3 succeeded at 692 with fuzz 1 (offset 20 lines).
Hunk #4 FAILED at 750.
Hunk #5 succeeded at 836 (offset 17 lines).
1 out of 5 hunks FAILED -- saving rejects to file asus-wmi.c.rej
```

- asus-wmi.c.rej :

```
--- asus-wmi.c	2020-05-25 15:53:24.136871000 +0200
+++ asus-wmi.c	2020-05-25 15:52:14.141040000 +0200
@@ -750,6 +816,8 @@ static void asus_wmi_led_exit(struct asu
 		led_classdev_unregister(&asus->wlan_led);
 	if (!IS_ERR_OR_NULL(asus->lightbar_led.dev))
 		led_classdev_unregister(&asus->lightbar_led);
+	if (!IS_ERR_OR_NULL(asus->screenpad_led.dev))
+		led_classdev_unregister(&asus->screenpad_led);
 	if (asus->led_workqueue)
 		destroy_workqueue(asus->led_workqueue);
 }
```

Plippo, 2021-01-27:
Looks like the script is using the wrong patchfile. It should use `patch5.8`, but from the output it seems as if it's using `patch`. I suppose there is an issue with the version check that I don't see, because for me it works. Can you execute the following command and post the output please?
`{ uname -r | grep -o '^[0-9]\+\.[0-9]\+' ; echo "5.7" ; } | sort -n`

sophietheopossum, 2021-06-15:
the same issue occurs on a fresh manjaro kde install. here's the output:
`[seirra@seirra-zenbookux482eaux482ea asus-wmi-1.0]$ sudo sh prepare-for-current-kernel.sh
Using: patch
--2021-06-15 21:24:37--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c?h=linux-5.10.y
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 136.144.49.103, 2604:1380:40b0:1a00::1
Connecting to git.kernel.org (git.kernel.org)|136.144.49.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 71104 (69K) [text/plain]
Saving to: ‘asus-wmi.c’

asus-wmi.c                   100%[==============================================>]  69.44K  --.-KB/s    in 0.06s   

2021-06-15 21:24:37 (1.11 MB/s) - ‘asus-wmi.c’ saved [71104/71104]

--2021-06-15 21:24:37--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.h?h=linux-5.10.y
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 136.144.49.103, 2604:1380:40b0:1a00::1
Connecting to git.kernel.org (git.kernel.org)|136.144.49.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2189 (2.1K) [text/plain]
Saving to: ‘asus-wmi.h’

asus-wmi.h                   100%[==============================================>]   2.14K  --.-KB/s    in 0s      

2021-06-15 21:24:37 (40.4 MB/s) - ‘asus-wmi.h’ saved [2189/2189]

--2021-06-15 21:24:37--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-nb-wmi.c?h=linux-5.10.y
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 136.144.49.103, 2604:1380:40b0:1a00::1
Connecting to git.kernel.org (git.kernel.org)|136.144.49.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17909 (17K) [text/plain]
Saving to: ‘asus-nb-wmi.c’

asus-nb-wmi.c                100%[==============================================>]  17.49K  --.-KB/s    in 0.02s   

2021-06-15 21:24:38 (876 KB/s) - ‘asus-nb-wmi.c’ saved [17909/17909]

patching file asus-nb-wmi.c
Hunk #1 succeeded at 578 with fuzz 1 (offset 109 lines).
Hunk #2 succeeded at 588 (offset 111 lines).
Hunk #3 succeeded at 602 (offset 111 lines).
patching file asus-wmi.c
Hunk #1 succeeded at 32 with fuzz 2.
Hunk #2 succeeded at 187 (offset 8 lines).
Hunk #3 succeeded at 692 with fuzz 1 (offset 20 lines).
Hunk #4 FAILED at 750.
Hunk #5 succeeded at 836 (offset 17 lines).
1 out of 5 hunks FAILED -- saving rejects to file asus-wmi.c.rej
[seirra@seirra-zenbookux482eaux482ea asus-wmi-1.0]$ { uname -r | grep -o '^[0-9]\+\.[0-9]\+' ; echo "5.7" ; } | sort -n
5.10
5.7`

i will proceed to try it out with the script set to use the 5.8 file and update with the result

Plippo, 2021-06-16:
Thank you for supplying the output! So I can confirm that it actually is a problem with the sorting. Now that I think about it it's clear that it can't work that way, I don't know why it sorted differently on my machine - probably a problem with the locale. I changed the script to use version sorting instead of numerical sorting and now it should work with all locales.

sophietheopossum, 2021-06-16:
tested and can confirm that it now correctly uses the newer patch and applies cleanly

kevinchappell, 2021-07-15:
@da-moon did [this change](https://github.com/Plippo/asus-wmi-screenpad/commit/657b43be15545bb132c66cf9a2a2b64dafd758ac) fix your issue? can this be closed now?

da-moon, 2021-07-15:
@kevinchappell I returned my zenbook so I won't be able to confirm whether this change has fixed the issue or not.

Plippo, 2021-07-26:
Closing as fix confirmed to work.

closed 2021-07-26 by Plippo