Plippo/asus-wmi-screenpad #17: DKMS Rebuild doesn't work after update (FEDORA)

jvaldenaire, 2021-02-08:
Hello ! 
I followed every step of you instruction, but i have to manually the step 5 commands after every major update of Fedora. Is this possible to make this rebuild automatic ?

Plippo, 2021-02-10:
It might be necessary to call `sudo dkms autoinstall` after a kernel update, can you give it a try?

jvaldenaire, 2021-02-10:
I get this, which might be the issue
```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.10.13-200.fc33.x86_64 all INCLUDEDIR=/lib/modules/5.10.13-200.fc33.x86_64/build/include KVERSION=5.10.13-200.fc33.x86_64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.13-200.fc33.x86_64 (x86_64)
Consult /var/lib/dkms/evdi/1.8.0/build/make.log for more information.

```

Plippo, 2021-02-10:
Has this been a major update, e.g from 5.8 to 5.10? Then there is no other way than uninstalling the module, run the script that downloads the sources for the new kernel and install it again. I will update the readme to make this clear. Otherwise, you can try looking at log file mentioned in your output.

jvaldenaire, 2021-02-11:
I will follow your instructions so, thanks. I'm still new to Linux and doesn't understand much of the logs, but i spotted this line : 
```
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";
```

Plippo, 2021-02-13:
I'm no Fedora expert, but have you installed the kernel headers and kernel devel for the correct version of your kernel? The error looks as if they are not installed. Maybe you have to install them manually after updating your kernel? I'm sorry there's not much I know about the kernel packages in Fedora...

jvaldenaire, 2021-02-14:
No problem, i will look how to ;)
Thanks a lot anyway for your answers !

closed 2021-02-14 by jvaldenaire