Plippo/asus-wmi-screenpad #18: Will this work on a UX482EA?

PUNJISTICK, 2021-03-28:
Hello!

I'm _extremely_ new to Linux and most of my time using Linux so far has been trying to learn how to adjust the backlight on my new UX482's ScreenPad. Primary monitor worked fine from day 1, backlight goes up/down with the keyboard keys just like on Windows. After scouring the internet I'm pretty sure you are the only human on this planet who seems to have actually worked it out for the ScreenPad, though. Your instructions seem very clear and I'm confident I can follow them, but I'm terrified of breaking everything and then having no idea what to do going forward.

That being said, before I go ahead and do it anyway, do you happen to know (or have a good guess) as to whether following your method here will work on my newer model? It's a UX482EA, not the UX481 as you did it on. I'm on Ubuntu Budgie if that matters, kernel version is 5.8.0-48-generic but I'm happy to change/upgrade that as needed.

I checked to see if the drivers matched between the older and newer model, but they do not. Same named drivers, but different versions. I'm by no means a coder and this whole world is very new to me, but I'm looking to learn!

Thanks for your time.

Plippo, 2021-04-02:
Hi,
of course I can't guarantee that ASUS hasn't changed the firmware interface, but I think it would not be very logical for them to do so. There isn't really a big chance to break anything by trying out the instructions, if it doesn't work it will either continue to use the old driver or you can just uninstall the modifed one and use the old one instead. DKMS only adds another version, it doesn't really replace the old driver. So I would encourage you to just try it out.
Cheers,
Philipp

PUNJISTICK, 2021-04-05:
Thanks a ton for the response, I was real worried about what damage I might cause screwing it up since this is all such an unknown to me. 

Wanted to come back and say it worked perfectly! I ended up doing it on an install of Parrot Linux, with this newer version laptop, works flawless. Thanks again.

closed 2021-04-05 by PUNJISTICK