Plippo/asus-wmi-screenpad #19: Manjaro - No such file or directory

Mawiel, 2021-03-30:
Hello, I am using Manjaro 21.0 with Kernel 5.12 and I get this error: 

cat: '/sys/class/leds/asus::screenpad/brightness': No such file or directory
/usr/bin/screenpad: line 5: [: =: unary operator expected
/usr/bin/screenpad: line 21: /sys/class/leds/asus::screenpad/brightness: No such file or directory

I tried both your commandlines and AUR package, same result, did I miss anything?

Cheers, Andreas :)

Plippo, 2021-04-02:
If the folder /sys/class/leds/asus::screenpad doesn't exist, the module either hasn't been installed correctly or doesn't recognize your device. Did you receive error messages installing the module? What device are you using?

sophietheopossum, 2021-06-15:
that may be related to #16 , also the current aur package will not correctly fail when dkms fails, so that may also be the cause

sophietheopossum, 2021-06-16:
hmmm after some more research, even with #16 resolved and using the script instructions to avoid the broken aur something still is not right. that's odd because #18 reported success on this device, the ux482ea. i'm  guessing there must be some sort of kernel difference, I will assume it's the version at fault, but if that fails i'll try to do some further investigation into whether it's a distribution specific patch

kevinchappell, 2021-07-25:
@sophietheopossum did you make any progress? I am now facing the same issue after a recent upgrade

MarsOrBust24, 2022-08-28:
This is a duplicate of #9 ... I got it working without any issue by installing from `yay -S asus-wmi-screenpad-dkms-git` On Manjaro Gnome 22 Linux 5.15 on an Asus UX8402