Plippo/asus-wmi-screenpad #21: Trying to install on Fedora 34 (beta) (same issue on Fedora 33 (stable)

legendhiro, 2021-04-04:
Newbie here. I managed to install the driver on Ubuntu and Pop_OS! 20.04/20.10 without any issues however, I am not able to do so on both Fedora 33 and 34.
Here are the return of the commands on the how to install guide:

$ sudo sh prepare-for-current-kernel.sh

Using: patch
--2021-04-04 23:07:35--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c?h=linux-5.11.y
Resolving git.kernel.org (git.kernel.org)... 2604:1380:40b0:1a00::1, 136.144.49.103
Connecting to git.kernel.org (git.kernel.org)|2604:1380:40b0:1a00::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 72487 (71K) [text/plain]
Saving to: ‘asus-wmi.c’

asus-wmi.c          100%[=================>]  70.79K  --.-KB/s    in 0.04s   

2021-04-04 23:07:35 (1.78 MB/s) - ‘asus-wmi.c’ saved [72487/72487]

--2021-04-04 23:07:35--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.h?h=linux-5.11.y
Resolving git.kernel.org (git.kernel.org)... 2604:1380:40b0:1a00::1, 136.144.49.103
Connecting to git.kernel.org (git.kernel.org)|2604:1380:40b0:1a00::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2215 (2.2K) [text/plain]
Saving to: ‘asus-wmi.h’

asus-wmi.h          100%[=================>]   2.16K  --.-KB/s    in 0s      

2021-04-04 23:07:35 (34.1 MB/s) - ‘asus-wmi.h’ saved [2215/2215]

--2021-04-04 23:07:35--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-nb-wmi.c?h=linux-5.11.y
Resolving git.kernel.org (git.kernel.org)... 2604:1380:40b0:1a00::1, 136.144.49.103
Connecting to git.kernel.org (git.kernel.org)|2604:1380:40b0:1a00::1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 18298 (18K) [text/plain]
Saving to: ‘asus-nb-wmi.c’

asus-nb-wmi.c       100%[=================>]  17.87K  --.-KB/s    in 0.02s   

2021-04-04 23:07:35 (969 KB/s) - ‘asus-nb-wmi.c’ saved [18298/18298]

prepare-for-current-kernel.sh: line 16: patch: command not found
rm: cannot remove '*.orig': No such file or directory

$ sudo dkms add -m asus-wmi -v 1.0

Creating symlink /var/lib/dkms/asus-wmi/1.0/source ->
                 /usr/src/asus-wmi-1.0

DKMS: add completed.

$ sudo dkms build -m asus-wmi -v 1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.11.11-300.fc34.x86_64 -C /lib/modules/5.11.11-300.fc34.x86_64/build M=/var/lib/dkms/asus-wmi/1.0/build...
cleaning build area...

DKMS: build completed.

$ sudo dkms install -m asus-wmi -v 1.0

asus-wmi.ko.xz:
Running module version sanity check.
 - Original module
   - Found /lib/modules/5.11.11-300.fc34.x86_64/extra/asus-wmi.ko.xz
   - Storing in /var/lib/dkms/asus-wmi/original_module/5.11.11-300.fc34.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.11.11-300.fc34.x86_64/extra/

asus-nb-wmi.ko.xz:
Running module version sanity check.
 - Original module
   - Found /lib/modules/5.11.11-300.fc34.x86_64/extra/asus-nb-wmi.ko.xz
   - Storing in /var/lib/dkms/asus-wmi/original_module/5.11.11-300.fc34.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.11.11-300.fc34.x86_64/extra/
Adding any weak-modules

depmod....

DKMS: install completed.

(* I rebooted my PC here *)

$ echo 255 | sudo tee '/sys/class/leds/asus::screenpad/brightness'

tee: '/sys/class/leds/asus::screenpad/brightness': No such file or directory
255

If anyone has an idea, I'd love to try it !

Plippo, 2021-04-06:
Well, it looks like the problem lies here:
```prepare-for-current-kernel.sh: line 16: patch: command not found```
So the files were not patched and thus you just reinstalled the same unmodified module you already had.
So you probably have to install the **patch** package first. I'm no Fedora expert, but I think you should be able to install the package using
```dnf install patch```
Then, perform all the steps again. I hope it works.

legendhiro, 2021-04-10:
This did the trick, thanks!

closed 2021-04-10 by legendhiro