Plippo/asus-wmi-screenpad #22: Get current screenpad brightness

AurelReb, 2021-04-07:
Currently when we try to get the current brightness with `cat /sys/class/leds/asus::screenpad/brightness` it always returns 0 instead of the current brightness.

Any way to implement this?

macbrownzie, 2021-04-07:
Good Morning AurelReb, 

Are you sure that you have permissions to read the file? 

On my Zenbook it works as expected. 
a@Zenbook:~$ cat /sys/class/leds/asus::screenpad/brightness
82
a@Zenbook:~$ cat /sys/class/leds/asus::screenpad/brightness
255

I did have to implement the chmod +x to enable file access. 

I hope that helps.

AurelReb, 2021-04-07:
Hi, thank you for your help.

here is the output of `ls -la` for the brightness file: 
`-rwxrwxrwx 1 root video 4.0K Apr  7 20:19 brightness`

When I run `cat` or `sudo cat`, the output is still the same: 0 even though the screenpad is at 255 brightness.

I'm on Arch Linux 5.10.27-1-lts if it can help.

AurelReb, 2021-04-07:
please note that `cat /sys/class/leds/asus::kbd_backlight/brightness` works great, the problem is specific to the screenpad

macbrownzie, 2021-04-07:
Are you able to control the brightness of the screenpad and turn it off (0) and on (>0) by "echo 0 > /sys/class/leds/asus::screenpad/brightness"  (IE:   Can you write to the file but and have the screenpad change state properly? 

There is another thread about Linux 5.10 kernels.   Have to reviewed it?

AurelReb, 2021-04-07:
Yes I can control brightness with `tee` and `light -s sysfs/leds/asus::screenpad` it works as expected.

I will check the 5.10 kernel thread

macbrownzie, 2021-04-08:
Have you tried "cat sysfs/leds/asus::screenpad"   If you are controlling the screedpad using that file my guess is that you need to read the light level from the same file  (at least that is how the drive works on my machine).

AurelReb, 2021-04-08:
Actually sysfs/leds/asus::screenpad is a path for `light` but corresponds to /sys/class/leds/asus::screenpad
But when I change the file manually, it changes the brightness but then when I read the file, the content is back to 0

MatthewCash, 2023-04-05:
I am having this same issue where the led classdev always returns 0.

It looks like calling `asus_wmi_get_devstate` on `ASUS_WMI_DEVID_SCREENPAD` returns a value that fails the check `(value & 0x1) == 0x1)`, which leads the driver to think the screen is deactivated.

When my screenpad is on, the return value is `10000000010100000` and off is `10000000000000000`. The activation check looks for the rightmost bit to be 1 but this is never the case on my machine. It seems like the activation check makes some assumptions that do not apply to all screenpads.

Removing this check fixes the brightness always returning 0, but now it will never return 0 if the screenpad is deactivated, instead showing the brightness as if it was still on. This makes sense because `screenpad_led_update` will never actually set the brightness to 0, instead it just turns off the screen.

Swapping the right side of the bitwise and with `0x20`, `0x80`, or both as `0xa0` (and then of course removing the `== 1`) completely fixed the problem for me, but I imagine it breaks it for the screenpads that function correctly with `0x1`.

Plippo, 2023-04-11:
That's quite interesting, thank you for investigating. On my model, the return values are 0x10001 when the screen is on and 0x10000 when it is off, that's why I added the check for 0x1. I guess changing the line from
`if (retval == 0 && (value & 0x1) == 0x1)`
to
`if (retval == 0 && (value & 0x21) != 0)`
(to check if either bit 0x1 or 0x20 is set) should make it work for both our machines. Of course we can't know if this has an impact for other models, but as it should not be very risky (in the worst case, the screenpad is reported as being on even if it is off) I would give it a go if it works for your model (I've just checked that it works for mine).

Could you try out if this works for you?

closed 2023-04-12 by Plippo