Plippo/asus-wmi-screenpad #23: The asus::screenpad folder isn't created

CptIceberg, 2021-05-06:
Hello ! 
I recently formated my hard drive, and installed Fedora 34 with the new Gnome40 desktop on my Asus Pro Duo.
It seems like the installation process isn't working : i have no errors whatsoever, but the asus::screenpad folder isn't created after restarting the computer. Any idea where does this come from  ?
Here's the return for the `{ uname -r | grep -o '^[0-9]\+\.[0-9]\+' ; echo "5.7" ; } | sort -n` command : 
```
5.11
5.7
```

CptIceberg, 2021-05-10:
To go deeper, here's the log from the prepare for kernel script : it seems it doesn't go all the way
```
Using: patch
--2021-05-10 09:01:34--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c?h=linux-5.11.y
Resolving git.kernel.org (git.kernel.org)... 136.144.49.103, 2604:1380:40b0:1a00::1
Connecting to git.kernel.org (git.kernel.org)|136.144.49.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 72487 (71K) [text/plain]
Saving to: ‘asus-wmi.c’

asus-wmi.c                              100%[===============================================================================>]  70.79K  --.-KB/s    in 0.03s

2021-05-10 09:01:35 (2.45 MB/s) - ‘asus-wmi.c’ saved [72487/72487]

--2021-05-10 09:01:35--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.h?h=linux-5.11.y
Resolving git.kernel.org (git.kernel.org)... 136.144.49.103, 2604:1380:40b0:1a00::1
Connecting to git.kernel.org (git.kernel.org)|136.144.49.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2215 (2.2K) [text/plain]
Saving to: ‘asus-wmi.h’

asus-wmi.h                              100%[===============================================================================>]   2.16K  --.-KB/s    in 0s

2021-05-10 09:01:35 (38.8 MB/s) - ‘asus-wmi.h’ saved [2215/2215]

--2021-05-10 09:01:35--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-nb-wmi.c?h=linux-5.11.y
Resolving git.kernel.org (git.kernel.org)... 136.144.49.103, 2604:1380:40b0:1a00::1
Connecting to git.kernel.org (git.kernel.org)|136.144.49.103|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 18298 (18K) [text/plain]
Saving to: ‘asus-nb-wmi.c’

asus-nb-wmi.c                           100%[===============================================================================>]  17.87K  --.-KB/s    in 0.01s

2021-05-10 09:01:35 (1.38 MB/s) - ‘asus-nb-wmi.c’ saved [18298/18298]

prepare-for-current-kernel.sh: line 16: patch: command not found
rm: cannot remove '*.orig': No such file or directory
```

closed 2021-05-12 by CptIceberg

SacDin, 2021-09-04:
what was the solution for this ?

CptIceberg, 2021-09-05:
It was a mistake from me, the "patch" shell command was not properly installed. A remove + reinstall did the trick