Plippo/asus-wmi-screenpad #25: /sys/class/leds/asus::screenpad/brightness not found with kernel 5.8.0-59-generic

simonchen007, 2021-07-11:
hello,
"/sys/class/leds/asus::screenpad/brightness" is not found with kernel 5.8.0-59-generic after installed.
I install is lib step by step, but is seems not works.
root@zenbook:/sys/class/leds# lsmod | grep "asus"
asus_nb_wmi            32768  0
asus_wmi               36864  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
wmi                    32768  2 asus_wmi,wmi_bmof
video                  49152  2 asus_wmi,i915
root@zenbook:/sys/class/leds# uname -a
Linux zenbook 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
root@zenbook:/sys/class/leds# cat /etc/os-release 
NAME="Linux Mint"
VERSION="20.2 (Uma)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20.2"
VERSION_ID="20.2"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=uma
UBUNTU_CODENAME=focal

Is there some suggestion for fix this issue ?

Thanks

closed 2021-07-14 by simonchen007

kevinchappell, 2021-07-25:
@simonchen007 Does your change in #26 fix the problem? I'm having the same issue even with `sudo depmod -a`

simonchen007, 2021-07-26:
@kevinchappell  , 
please try the fllowing:
```bash
cd /lib/modules/YOURKERNELVERSION/kernel/drivers/platform/x86
mv  asus-nb-wmi.ko asus-nb-wmi.ko_bak
mv asus-wmi.ko asus-wmi.ko_bak
ln -s ../../../../extra/asus-nb-wmi.ko ./
ln -s ../../../../extra/asus-wmi.ko ./
```
It may works after reboot.

KeenanLawrence, 2021-07-26:
@simonchen007 
I tried that and still not working after reboot.

P.S It was working prior to the kernel update. My kernel version is `5.8.0.63-generic`

simonchen007, 2021-07-26:
@KeenanLawrence 
I'm using the same version of Kernel.
Please check it.
```bash
root@zenbook:/lib/modules/5.8.0-63-generic/kernel/drivers/platform/x86# ls -la asus*
-rw-r--r-- 1 root root 50673 Jul 15 21:51 asus-laptop.ko
lrwxrwxrwx 1 root root    32 Jul 21 21:34 asus-nb-wmi.ko -> ../../../../extra/asus-nb-wmi.ko
-rw-r--r-- 1 root root 31673 Jul 15 21:51 asus-nb-wmi_ko
-rw-r--r-- 1 root root 13993 Jul 15 21:51 asus-wireless.ko
lrwxrwxrwx 1 root root    29 Jul 21 21:34 asus-wmi.ko -> ../../../../extra/asus-wmi.ko
-rw-r--r-- 1 root root 60169 Jul 15 21:51 asus-wmi_ko
root@zenbook:/lib/modules/5.8.0-63-generic/kernel/drivers/platform/x86# cd ../../../../extra/
root@zenbook:/lib/modules/5.8.0-63-generic/extra# ls -la
total 100
drwxr-xr-x 2 root root  4096 Jul 21 21:30 .
drwxr-xr-x 7 root root  4096 Jul 21 21:30 ..
-rw-r--r-- 1 root root 31307 Jul 21 21:30 asus-nb-wmi.ko
-rw-r--r-- 1 root root 60947 Jul 21 21:30 asus-wmi.ko
root@zenbook:/lib/modules/5.8.0-63-generic/extra# 
```
it works after reboot in myside.