Plippo/asus-wmi-screenpad #29: /sys/class/leds/asus::screenpad not found (zephyrus duo / fedora)

anonsquest, 2021-09-07:
Any chance I could get it working with zeph duo se variant?
I have already tried the troubleshooting command. 
Also tried `sudo mokutil --disable-validation` as well and actually disabled SB. 
Both did not work

`uname -r`
`5.14.0-60.rog.fc34.x86_64'`

`sudo sh prepare-for-current-kernel.sh`

`Using: patch5.8
--2021-09-06 21:22:01--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c?h=linux-5.14.y
Resolving git.kernel.org (git.kernel.org)... 139.178.84.217, 2604:1380:4641:c500::1
Connecting to git.kernel.org (git.kernel.org)|139.178.84.217|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 72575 (71K) [text/plain]
Saving to: ‘asus-wmi.c’

asus-wmi.c          100%[===================>]  70.87K   175KB/s    in 0.4s    

2021-09-06 21:22:03 (175 KB/s) - ‘asus-wmi.c’ saved [72575/72575]

--2021-09-06 21:22:03--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.h?h=linux-5.14.y
Resolving git.kernel.org (git.kernel.org)... 139.178.84.217, 2604:1380:4641:c500::1
Connecting to git.kernel.org (git.kernel.org)|139.178.84.217|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2215 (2.2K) [text/plain]
Saving to: ‘asus-wmi.h’

asus-wmi.h          100%[===================>]   2.16K  --.-KB/s    in 0s      

2021-09-06 21:22:04 (18.1 MB/s) - ‘asus-wmi.h’ saved [2215/2215]

--2021-09-06 21:22:04--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-nb-wmi.c?h=linux-5.14.y
Resolving git.kernel.org (git.kernel.org)... 139.178.84.217, 2604:1380:4641:c500::1
Connecting to git.kernel.org (git.kernel.org)|139.178.84.217|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16940 (17K) [text/plain]
Saving to: ‘asus-nb-wmi.c’

asus-nb-wmi.c                 100%[===============================================>]  16.54K  88.5KB/s    in 0.2s    

2021-09-06 21:22:05 (88.5 KB/s) - ‘asus-nb-wmi.c’ saved [16940/16940]

patching file asus-nb-wmi.c
Hunk #1 succeeded at 544 (offset -34 lines).
Hunk #2 succeeded at 554 (offset -34 lines).
Hunk #3 succeeded at 568 (offset -34 lines).
patching file asus-wmi.c
Hunk #2 succeeded at 191 (offset 4 lines).
Hunk #3 succeeded at 722 (offset 30 lines).
Hunk #4 succeeded at 867 (offset 30 lines).`

`sudo dkms add -m asus-wmi -v 1.0`

`Creating symlink /var/lib/dkms/asus-wmi/1.0/source ->
                 /usr/src/asus-wmi-1.0

DKMS: add completed.`

`sudo dkms build -m asus-wmi -v 1.0`

`
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.14.0-60.rog.fc34.x86_64 -C /lib/modules/5.14.0-60.rog.fc34.x86_64/build M=/var/lib/dkms/asus-wmi/1.0/build...
cleaning build area...

DKMS: build completed.`

`sudo dkms install -m asus-wmi -v 1.0`

`Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.14.0-60.rog.fc34.x86_64 -C /lib/modules/5.14.0-60.rog.fc34.x86_64/build M=/var/lib/dkms/asus-wmi/1.0/build...
cleaning build area...

DKMS: build completed.
[an0n@fedora asus-wmi-1.0]$ sudo dkms install -m asus-wmi -v 1.0

asus-wmi.ko.xz:
Running module version sanity check.
modinfo: ERROR: Module /lib/modules/5.14.0-60.rog.fc34.x86_64/kernel/drivers/platform/x86/asus-wmi.ko not found.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.14.0-60.rog.fc34.x86_64/extra/

asus-nb-wmi.ko.xz:
Running module version sanity check.
modinfo: ERROR: Module /lib/modules/5.14.0-60.rog.fc34.x86_64/kernel/drivers/platform/x86/asus-nb-wmi.ko not found.
 - Original module
   - Found /lib/modules/5.14.0-60.rog.fc34.x86_64/extra/asus-nb-wmi.ko.xz
   - Storing in /var/lib/dkms/asus-wmi/original_module/5.14.0-60.rog.fc34.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/5.14.0-60.rog.fc34.x86_64/extra/

depmod....

DKMS: install completed.`

Z-100, 2021-09-07:
I have the same problem on Ubuntu 20.04 on Kernel 5.11.
It is building without problems but the folder does not exist after a reboot.

chowdri, 2021-09-17:
Same problem sing kernel upgrade to 5.11.0-34 on ubuntu(elementary).

OggyP, 2021-09-17:
I am also having this issue on Ubuntu 21.04 Kernel 5.11 UX481FL. Worked on previous kernels. 
`tee: '/sys/class/leds/asus::screenpad/brightness': No such file or directory`

chowdri, 2021-09-17:
I think the troubleshooting edits work, sometimes.

OggyP, 2021-09-18:
I can not run the troubleshooting steps. When I try the troubleshooting steps after installing and rebooting
`cd /lib/modules/5.11.0-34-generic/kernel/drivers/platform/x86                                      
sudo mv  asus-nb-wmi.ko asus-nb-wmi.ko_bak
sudo mv asus-wmi.ko asus-wmi.ko_bak
sudo ln -s ../../../../extra/asus-nb-wmi.ko ./
sudo ln -s ../../../../extra/asus-wmi.ko ./
sudo depmod -a`
it gives the errors
`depmod: ERROR: fstatat(7, asus-wmi.ko_bak): No such file or directory
depmod: ERROR: fstatat(7, asus-nb-wmi.ko): No such file or directory
depmod: ERROR: fstatat(7, asus-nb-wmi.ko_bak): No such file or directory
depmod: ERROR: fstatat(7, asus-wmi.ko): No such file or directory`

chowdri, 2021-09-23:
> I think the troubleshooting edits work, sometimes.

On UX581GV, the troubleshooting edits work when upon kernel update you uninstall nvidia drivers, then reinstall screenpad drivers. Once screenpad control works, reinstall nvidia drivers. 
In lshw, the screenpad seems to be listed under the nvidia pci device.

ashwin-M-D, 2021-09-28:
@chowdri Any way you could explain how you got it to work in more detail? As for the Nvidia drivers, they technically should not interfere with the location of the screenpad (device file in Linux). Besides, this negates the point that it works in older versions of the kernel but doesn't in the newer versions.
I see dkms successfully building and installing the module but the device doesn't show up in the leds directory. I am currently running kernel 5.13. I had it working without any problems up until 5.10. kernel 5.11 is when it broke down.

chowdri, 2021-09-29:
on the question of why nvidia drivers interfere in the function, I don't know for sure since the driver and process sequencing are not obvious. I stumbled upon it through trial and error, and the only difference I could ascertain was shown in lshw. 

(without nvidia driver)
       *-display
            description: VGA compatible controller
            product: TU106M [GeForce RTX 2060 Mobile]
            vendor: NVIDIA Corporation
            physical id: 0
            bus info: pci@0000:01:00.0
            logical name: /dev/fb0
            version: a1
            width: 64 bits
            clock: 33MHz
            capabilities: pm msi pciexpress vga_controller bus_master cap_list rom fb
            configuration: depth=32 driver=nouveau latency=0 mode=3840x1100 visual=truecolor xres=3840 yres=1100
            resources: iomemory:600-5ff iomemory:600-5ff irq:164 memory:83000000-83ffffff memory:6030000000-603fffffff memory:6040000000-6041ffffff ioport:3000(size=128) memory:84000000-8407ffff 

(with nvidia driver)
       *-display
            description: VGA compatible controller
            product: TU106M [GeForce RTX 2060 Mobile]
            vendor: NVIDIA Corporation
            physical id: 0
            bus info: pci@0000:01:00.0
            logical name: /dev/fb0
            version: a1
            width: 64 bits
            clock: 33MHz
            capabilities: pm msi pciexpress vga_controller bus_master cap_list fb
            configuration: depth=32 latency=0 mode=3840x1100 visual=truecolor xres=3840 yres=1100
            resources: iomemory:600-5ff iomemory:600-5ff memory:83000000-83ffffff memory:6030000000-603fffffff memory:6040000000-6041ffffff ioport:3000(size=128) memory:84000000-8407ffff

Also, the troubleshooting steps need to be executed before reinstalling the nvidia drivers.
One has to check the driver module dependencies to grasp the the nature of problem.

anonsquest, 2021-10-01:
I switched to arch linux and that fixed the issue.

closed 2021-10-01 by anonsquest

frees11, 2025-04-25:
This is definitely not solved, I have same problem on latest lts ubuntu with Zephyrus DUO GX650RX