Plippo/asus-wmi-screenpad #35: WARNING: `dkms install --no-depmod asus-wmi/1.0 -k <other kernel>' exited 10

HaleTom, 2022-01-17:
Note It works on my current kernel (5.15) but fails on 5.4 and 5.9:

```
ravi@svelte:~/.cache/aur/asus-wmi-screenpad-dkms-git (master*% u=)% makepkg  -sfi                                                                                                        1m23s | 22-01-17 19:06:06
==> Making package: asus-wmi-screenpad-dkms-git r30.47fce74-1 (Mon 17 Jan 2022 19:09:41)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating asus-wmi-screenpad-dkms-git git repo...
Fetching origin
==> Validating source files with sha512sums...
    asus-wmi-screenpad-dkms-git ... Skipped
==> Extracting sources...
  -> Creating working copy of asus-wmi-screenpad-dkms-git git repo...
Reset branch 'makepkg'
==> Starting prepare()...
Using: patch5.8
--2022-01-17 19:09:48--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c?h=linux-5.15.y
SSL_INIT
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 145.40.68.75, 2604:1380:4601:e00::1
Connecting to git.kernel.org (git.kernel.org)|145.40.68.75|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 81873 (80K) [text/plain]
Saving to: ‘asus-wmi.c’

asus-wmi.c                                           100%[=====================================================================================================================>]  79.95K   113KB/s    in 0.7s

2022-01-17 19:09:55 (113 KB/s) - ‘asus-wmi.c’ saved [81873/81873]

--2022-01-17 19:09:55--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.h?h=linux-5.15.y
SSL_INIT
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 145.40.68.75, 2604:1380:4601:e00::1
Connecting to git.kernel.org (git.kernel.org)|145.40.68.75|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2215 (2.2K) [text/plain]
Saving to: ‘asus-wmi.h’

asus-wmi.h                                           100%[=====================================================================================================================>]   2.16K  --.-KB/s    in 0s

2022-01-17 19:10:01 (46.7 MB/s) - ‘asus-wmi.h’ saved [2215/2215]

--2022-01-17 19:10:01--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-nb-wmi.c?h=linux-5.15.y
SSL_INIT
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 145.40.68.75, 2604:1380:4601:e00::1
Connecting to git.kernel.org (git.kernel.org)|145.40.68.75|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16940 (17K) [text/plain]
Saving to: ‘asus-nb-wmi.c’

asus-nb-wmi.c                                        100%[=====================================================================================================================>]  16.54K  79.2KB/s    in 0.2s

2022-01-17 19:10:08 (79.2 KB/s) - ‘asus-nb-wmi.c’ saved [16940/16940]

patching file asus-nb-wmi.c
Hunk #1 succeeded at 544 (offset -34 lines).
Hunk #2 succeeded at 554 (offset -34 lines).
Hunk #3 succeeded at 568 (offset -34 lines).
patching file asus-wmi.c
Hunk #1 succeeded at 33 (offset 1 line).
Hunk #2 succeeded at 192 (offset 5 lines).
Hunk #3 succeeded at 910 (offset 218 lines).
Hunk #4 succeeded at 1055 (offset 218 lines).
==> Starting pkgver()...
==> Removing existing $pkgdir/ directory...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "asus-wmi-screenpad-dkms-git"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: asus-wmi-screenpad-dkms-git r30.47fce74-1 (Mon 17 Jan 2022 19:10:10)
==> Installing package asus-wmi-screenpad-dkms-git with /usr/bin/powerpill -U...
loading packages...
warning: asus-wmi-screenpad-dkms-git-r30.47fce74-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) asus-wmi-screenpad-dkms-git-r30.47fce74-1

Total Installed Size:  0.13 MiB
Net Upgrade Size:      0.07 MiB

:: Proceed with installation? [Y/n]
(1/1) checking keys in keyring                                                                                                 [-----------------------------------------------------------------------------] 100%
(1/1) checking package integrity                                                                                               [-----------------------------------------------------------------------------] 100%
(1/1) loading package files                                                                                                    [-----------------------------------------------------------------------------] 100%
(1/1) checking for file conflicts                                                                                              [-----------------------------------------------------------------------------] 100%
(1/1) checking available disk space                                                                                            [-----------------------------------------------------------------------------] 100%
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove asus-wmi/1.0
:: Processing package changes...
(1/1) reinstalling asus-wmi-screenpad-dkms-git                                                                                 [-----------------------------------------------------------------------------] 100%
:: Running post-transaction hooks...
(1/4) Arming ConditionNeedsUpdate...
(2/4) Install DKMS modules
==> dkms install --no-depmod asus-wmi/1.0 -k 5.4.169-1-MANJARO
Error! Bad return status for module build on kernel: 5.4.169-1-MANJARO (x86_64)
Consult /var/lib/dkms/asus-wmi/1.0/build/make.log for more information.
==> WARNING: `dkms install --no-depmod asus-wmi/1.0 -k 5.4.169-1-MANJARO' exited 10
==> dkms install --no-depmod asus-wmi/1.0 -k 5.10.89-1-MANJARO
Error! Bad return status for module build on kernel: 5.10.89-1-MANJARO (x86_64)
Consult /var/lib/dkms/asus-wmi/1.0/build/make.log for more information.
==> WARNING: `dkms install --no-depmod asus-wmi/1.0 -k 5.10.89-1-MANJARO' exited 10
==> dkms install --no-depmod asus-wmi/1.0 -k 5.15.12-1-MANJARO
==> depmod 5.15.12-1-MANJARO
(3/4) Cleaning pacman cache...
Installed packages:
==> no candidate packages found for pruning
Uninstalled packages:
==> no candidate packages found for pruning
(4/4) Checking which packages need to be rebuilt
```

I see the file `prepare-for-current-kernel.sh`.  How can I have this work properly for non-current kernels, too?

HaleTom, 2022-01-17:
The only `make.log` left on my system is for kernel 5.15 which shows everything worked just fine.

HaleTom, 2022-01-18:
I think the problem is that the module is built only for one kernel, so trying to install it on other kernels won't work:

Here, 5.15 works, 5.10 fails:

```
% sudo dkms -k 5.15.12-1-MANJARO --kernelsourcedir /usr/lib/modules/5.15.12-1-MANJARO/build install asus-wmi/1.0

Module asus-wmi/1.0 already installed on kernel 5.15.12-1-MANJARO (x86_64).
% sudo dkms -k 5.10.89-1-MANJARO --kernelsourcedir /usr/lib/modules/5.10.89-1-MANJARO/build install asus-wmi/1.0

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.10.89-1-MANJARO -C /usr/lib/modules/5.10.89-1-MANJARO/build M=/var/lib/dkms/asus-wmi/1.0/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.89-1-MANJARO (x86_64)
Consult /var/lib/dkms/asus-wmi/1.0/build/make.log for more information.
% cat /var/lib/dkms/asus-wmi/1.0/build/make.log
DKMS make.log for asus-wmi-1.0 for kernel 5.10.89-1-MANJARO (x86_64)
Tue 18 Jan 2022 17:21:02 +07
make: Entering directory '/usr/lib/modules/5.10.89-1-MANJARO/build'
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-nb-wmi.o
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:29:10: fatal error: linux/platform_profile.h: No such file or directory
   29 | #include <linux/platform_profile.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:280: /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1822: /var/lib/dkms/asus-wmi/1.0/build] Error 2
make: Leaving directory '/usr/lib/modules/5.10.89-1-MANJARO/build'
%
```

`platform_profile.h` only exists in 5.15:

```
% curl -s https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c\?h\=linux-5.15.y | grep -F 'platform_profile.h'
#include <linux/platform_profile.h>
% fd platform_profile.h /usr/lib/modules/
/usr/lib/modules/5.15.12-1-MANJARO/build/include/linux/platform_profile.h
```
It doesn't exist in 5.10:
```
% fd platform_profile.h /usr/lib/modules/5.10.89-1-MANJARO
%
```

Could you build for all installed kernels, or allow passing a kernel name to build for?

jibsaramnim, 2022-01-19:
You could try to temporarily hard-code the specific Kernel version you'd like to build and install this for in line two of `prepare-for-current-kernel.sh` (https://github.com/Plippo/asus-wmi-screenpad/blob/47fce74f278e3896d2ca5bdcfc8f5592d3ca028a/prepare-for-current-kernel.sh#L2), something like replacing it with `VERSION="5.10"` with that specific Kernel version actually already installed on your system might do the trick. 

Note that I haven't tried this myself, but if you do end up trying it, I'd love to hear if it worked. If it does work, it should be a relatively simple change to have the `VERSION` value either be provided (as an environment variable or command argument), or alternatively fall back to the currently running version (ie. the current behavior).

jibsaramnim, 2022-02-09:
I just wanted to follow up my previous post and to let you know that it indeed works as you want. The Linux flavor I'm running (Fedora 35) just updated to kernel 5.16.5 from 5.15.7 (I believe), and I was able to install the new kernel and immediately update and install the screenpad module, before rebooting. The steps I took:

- Manually modified `prepare-for-current-kernel.sh` so that it had `VERSION="5.16"` at the very top of the file
- Ran `sudo sh prepare-for-current-kernel.sh`
- Ran dkms build with the new kernel version specified, in my case: `sudo dkms build -m asus-wmi -v 1.0 -k 5.16.5-200.fc35.x86_64`
- Ran dkms install with the new kernel specified, in my case: `sudo dkms install -m asus-wmi -v 1.0 -k 5.16.5-200.fc35.x86_64`
- Reboot

It should be possibly to modify the prepare shell script to optionally take a version number from the command line so that it doesn't have to be hard-coded, but as I'm a little pressed foro time right now, simply temporarily setting it like this does the trick.

I hope this helps!