Plippo/asus-wmi-screenpad #38: Linux 5.17.5  '/sys/class/leds/asus::screenpad': No such file or directory

fabian-milea, 2022-06-16:
After having the module up and running for a while, with the latest kernel upgrade I can not get it working again... tried the remove and reinstall several times... Anyone having issues on Kernel > 5.17?

`mcf@:aws$ uname -a
Linux mcf 5.17.5-76051705-generic #202204271406~1653440576~22.04~6277a18 SMP PREEMPT Wed May 25 01 x86_64 x86_64 x86_64 GNU/Linux
`
` lsmod | grep asus
asus_nb_wmi            28672  0
asus_wmi               53248  1 asus_nb_wmi
platform_profile       16384  1 asus_wmi
sparse_keymap          16384  1 asus_wmi
wmi                    32768  2 asus_wmi,wmi_bmof
video                  53248  2 asus_wmi,i915
`

jibsaramnim, 2022-06-17:
I've been running things without issue on my Zenbook Duo 14 running Fedora 36, even on Kernel >=5.17. 

There was someone else who reported having some challenges under Arch who was able to make it work by adjusting the Kernel module loading order, as somehow one of the modules was getting in the way of the asus_wmi one. They ended up making `asus_wmi` to load first by adding it to `mkinicpio.conf`, they discovered this after looking into what `journalctl -b` reported.

I am not sure what flavor you're running, but if you're on Ubuntu or on of its derivatives, be sure to check the [troubleshooting](https://github.com/Plippo/asus-wmi-screenpad#troubleshooting) section if you haven't already -- specifically the second half that mentions the `mdf_aaeon` extension and how to disable that.

jibsaramnim, 2022-06-17:
_Moved [my comment to the other issue](https://github.com/Plippo/asus-wmi-screenpad/issues/37#issuecomment-1158551285) as I realized you posted there too_

fabian-milea, 2022-06-17:
You are right I had to run the commands in the troubleshooting to fully remove the module before a new install:

sudo mv  asus-nb-wmi.ko asus-nb-wmi.ko_bak
sudo mv asus-wmi.ko asus-wmi.ko_bak
sudo ln -s ../../../../extra/asus-nb-wmi.ko ./
sudo ln -s ../../../../extra/asus-wmi.ko ./
sudo depmod -a

jibsaramnim, 2022-06-17:
Very nice, I'm glad you got it to work! Be sure to close this issue as it's fully resolved now then :).

closed 2022-06-17 by fabian-milea