Plippo/asus-wmi-screenpad #39: Spontaneously stopped working [Arch]

lyiriyah, 2022-07-18:
`/sys/class/leds/asus::screenpad/brightness` exists, but writing to it has no effect. Occurs on 5.18.12 and 5.18.10. The `/lib/modules/5.18.{10,12}/extra` folders are empty, which is odd...

`lsmod | grep asus`: ```
asus_nb_wmi            28672  0
asus_wmi               73728  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
platform_profile       16384  1 asus_wmi
rfkill                 36864  7 iwlmvm,asus_wmi,bluetooth,cfg80211
video                  57344  2 asus_wmi,i915
wmi                    45056  3 asus_wmi,wmi_bmof,mxm_wmi
i8042                  45056  1 asus_nb_wmi
```
`dmesg | grep asus`: ```
[    2.036322] asus_wmi: ASUS WMI generic driver loaded
[    2.038037] asus_wmi: Initialization: 0x1
[    2.038103] asus_wmi: BIOS WMI version: 9.0
[    2.038146] asus_wmi: SFUN value: 0x21
[    2.038149] asus-nb-wmi asus-nb-wmi: Detected ATK, not ASUSWMI, use DSTS
[    2.038152] asus-nb-wmi asus-nb-wmi: Detected ATK, enable event queue
[    2.042006] input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input26
```

Thanks in advance for any support you can offer.

lyiriyah, 2022-07-18:
I switched to the fallback initramfs image on 5.18.12 and rebooted a few times -- that seemed to work. Don't know what the actual fix is.

closed 2022-07-18 by lyiriyah

MarsOrBust24, 2022-08-28:
Did the acpi_call-dkms method work?

To try it run:

```bash
sudo pacman -Syu linux518-headers
pamac install acpi_call-dkms
echo '\_SB.ATKD.WMNB 0x0 0x53564544 b32000500FF000000' | sudo tee /proc/acpi/call
```

That sets the brightness to 0xFF

lyiriyah, 2022-09-25:
Sorry, didn't see this for a while
I can check for you in a couple hours