Plippo/asus-wmi-screenpad #42: Lost keyboard backlight, no file appears in Leds directory.

Jrms13, 2022-10-16:
I tried the commands mention in the page but when I looked into the Leds directory, theres no Screenpad item in there. Also if soneone could point me in the right direction to install the dkms and linux headers on fedora  36. Im new and this is my first linux install.

jibsaramnim, 2022-10-18:
I'm not the creator of the kernel extension, but could you share what steps you have taken (as exact as possible), and what specific model of Zenbook laptop you have?

Jrms13, 2022-10-22:
I followed the directions in the read me , currently using Fedora workstation 36. In step 5, `sudo dkms build -m asus-wmi -v 1.0
sudo dkms install -m asus-wmi -v 1.0` keep getting an error, file not found. I don't want to try again, since I use the keyboard backlite a lot and im not sure how to fix it if i try it again. Still learning my way around linux. But it certainly would be nice to increase the brightness of the second screen.

jibsaramnim, 2022-10-24:
> `sudo dkms build -m asus-wmi -v 1.0 sudo dkms install -m asus-wmi -v 1.0` keep getting an error, file not found. 

I am not sure if this was a copy/paste oopsie in your comment, but those are two separate commands and should be run individually; `sudo dkms build -m asus-wmi -v 1.0` and `sudo dkms install -m asus-wmi -v 1.0`.

Beyond that, can you share the full results if the command(s) still fail? As in, whatever the command returns, please share it here so we can see the full context and, hopefully, what file it might be missing.

fabian-milea, 2022-11-28:
are you running Kernel version 6.0? I believe that since upgrading to kernel 6 I also lost keyboard light and also second screen brightness control

Knoxi-Code, 2022-12-31:
Hey i have lost my keybordlight too :(

and icant build and install the module to the current kernel:

`#sudo dkms build -m asus-wmi -v 1.0
Building module:
cleaning build area...
make -j8 KERNELRELEASE=6.0.0-2parrot1-amd64 -C /lib/modules/6.0.0-2parrot1-amd64/build M=/var/lib/dkms/asus-wmi/1.0/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.0.0-2parrot1-amd64 (x86_64)
Consult /var/lib/dkms/asus-wmi/1.0/build/make.log for more information.
`
What can i do ?