Plippo/asus-wmi-screenpad #47: DKMS aborting installation when running the aur package on Garuda linux w/ the asus ctl kernel.

orbitaldr0p, 2023-01-30:
DKMS is refusing the install the new kernel because it exactly matches some that already exists, and causes the AUR package to not work. Following the instuctions in the readme doesn't work either. asus::screenpad doesnt appear.

orbitaldr0p, 2023-01-30:
I removed asusctl's kernel and DKMS still refuses to install.

PTFOPlayer, 2023-01-31:
use --force when using dkms
that will help

orbitaldr0p, 2023-01-31:
how do I do that with the DKMS that is run when installing with yay?

PTFOPlayer, 2023-02-01:
unfortunately you have to do the installation manually

boscowitch, 2023-04-28:
Unfortunately this keeps happening with every kernel update, wmi stops working and dkms complains and a manual install (or modification of the aur script) is needed.

But what is the actuall issue though, why does this start happening, I don't know much about kernel modules, is there something missing in the definition for newer kernels for dkms to continue working, if so cant we add this, since it seems to work fine after force, just not recompiling on kernel updates (which happen often in Archlinux...).

boscowitch, 2023-04-28:
BTW another way on my system to reinstall instead of force was.
1. do the dkms remove like in the readme
2. uninstall the aur package (or manually remove the folder like in the readme)
3. search the kernel dir for the asus wmi ko file that reinstall compains about to be present in the given kernel number, for me atm it was: /lib/modules/6.2.12-arch1-1/kernel/drivers/platform/x86/asus-nb-wmi.ko.zst  and /lib/modules/6.2.12-arch1-1/kernel/drivers/platform/x86/asus-wmi.ko.zst and remove them,
4. reinstall aur package

(i had another unrelated dkms error since I hat also linux lts kernel header package installed but not the kernel, i just had to remove that too, but normally this should not be a problem if you havent installed those headers by accident)