Plippo/asus-wmi-screenpad #48: Error building for Kernel 6.2.2-arch1-1

Kwanky-Millet, 2023-03-04:
The module build returns bad exit status: 2

**Make Log FIle**
`DKMS make.log for asus-wmi-1.0 for kernel 6.2.2-arch1-1 (x86_64)
Saturday 04 March 2023 08:23:15 PM IST
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-nb-wmi.o
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:925:24: error: initialization of ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ from incompatible pointer type ‘int (*)(struct power_supply *)’ [-Werror=incompatible-pointer-types]
  925 |         .add_battery = asus_wmi_battery_add,
      |                        ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:925:24: note: (near initialization for ‘battery_hook.add_battery’)
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:926:27: error: initialization of ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ from incompatible pointer type ‘int (*)(struct power_supply *)’ [-Werror=incompatible-pointer-types]
  926 |         .remove_battery = asus_wmi_battery_remove,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:926:27: note: (near initialization for ‘battery_hook.remove_battery’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o] Error 1
make: *** [Makefile:2021: /var/lib/dkms/asus-wmi/1.0/build] Error 2`

GeraldWodni, 2023-03-09:
Can confirm, here is my log:

```
DKMS make.log for asus-wmi-1.0 for kernel 6.2.2-arch1-1 (x86_64)
Thu  9 Mar 17:35:54 CET 2023
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-nb-wmi.o
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:932:24: error: initialization of ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ from incompatible pointer type ‘int (*)(struct power_supply *)’ [-Werror=incompatible-pointer-types]
  932 |         .add_battery = asus_wmi_battery_add,
      |                        ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:932:24: note: (near initialization for ‘battery_hook.add_battery’)
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:933:27: error: initialization of ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ from incompatible pointer type ‘int (*)(struct power_supply *)’ [-Werror=incompatible-pointer-types]
  933 |         .remove_battery = asus_wmi_battery_remove,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:933:27: note: (near initialization for ‘battery_hook.remove_battery’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o] Error 1
make: *** [Makefile:2021: /var/lib/dkms/asus-wmi/1.0/build] Error 2

```

GeraldWodni, 2023-03-20:
It seems the linux [ABI has changed in version 6.2](https://github.com/torvalds/linux/commit/878a82c23469d6626b27b44ade4d8f9438de51ab)

There must be a lot of affected modules out there. I found one that had the same error and was [fixed](https://bugs.launchpad.net/ubuntu/+source/librem-ec-acpi/+bug/2009511)

The [patch shows](https://launchpadlibrarian.net/654770142/librem-ec-acpi-support-linux-6.2.debdiff), that it might not be hard to fix.

I might try and have a go at it, or do you @Kwanky-Millet want to give it a shot?

closed 2023-03-21 by Kwanky-Millet