Plippo/asus-wmi-screenpad #50: Does not work on Ubuntu 22.10

ExposedCat, 2023-03-13:
OS: Ubuntu 22.10 x86_64
Host: ZenBook UX482EA_UX482EA 1.0
Kernel: 5.19.0-35-generic
GPU: Intel TigerLake-LP GT2 [Iris Xe Graphics]

```
[  +0.002556] asus_wmi: ASUS WMI generic driver loaded
[  +0.018890] asus_nb_wmi: loading out-of-tree module taints kernel.
[  +0.000073] asus_nb_wmi: module verification failed: signature and/or required key missing - tainting kernel
[  +0.003771] asus_wmi: Initialization: 0x1
[  +0.002063] asus_wmi: BIOS WMI version: 9.0
[  +0.000114] asus_wmi: SFUN value: 0x21
[  +0.000007] asus-nb-wmi asus-nb-wmi: Detected ATK, not ASUSWMI, use DSTS
[  +0.000006] asus-nb-wmi asus-nb-wmi: Detected ATK, enable event queue
[  +0.009751] input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input23
[  +0.009166] asus_wmi: fan_curve_get_factory_default (0x00110024) failed: -19
[  +0.000635] asus_wmi: fan_curve_get_factory_default (0x00110025) failed: -19
```

MDCore, 2023-03-20:
I had a similar problem with it not working on 22.10. I fixed it by turning off Safe Boot in the BIOS.

PTFOPlayer, 2023-03-20:
consider installing and  building with --force flag

ExposedCat, 2023-03-23:
@PTFOPlayer `--force` did not help
@MDCore turned off for me 🤔

MDCore, 2023-03-24:
@ExposedCat I turned off then uninstalled and reinstalled - including the extra blacklisting step. Beyond that, I don't have any suggestions unfortunately!

lamualfa, 2023-09-07:
1. Make sure to turned off the **Secure Boot**
2. Run the following command:

```bash
echo "blacklist mdf_aaeon" | sudo tee /etc/modprobe.d/aaeon-blacklist.conf
sudo update-initramfs -k all -u
```

3. Uninstall the kernel module:

```bash
sudo dkms remove -m asus-wmi -v 1.0 --all
sudo rm -r /usr/src/asus-wmi-1.0
```

4. Restart your computer
5. Repeat the **Step 2** - **Step 5** in [README.md](https://github.com/Plippo/asus-wmi-screenpad)