Plippo/asus-wmi-screenpad #51: Bad return status for module build on kernel: 6.0.7-301.fc37.x86_64 (x86_64)

Fir3Storm31, 2023-03-14:
### System info

**uname -r:** 
```
6.0.7-301.fc37.x86_64
```

**cat /etc/os-release:**
```
NAME="Fedora Linux"
VERSION="37 (Workstation Edition)"
ID=fedora
VERSION_ID=37
VERSION_CODENAME=""
PLATFORM_ID="platform:f37"
PRETTY_NAME="Fedora Linux 37 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:37"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f37/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=37
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=37
SUPPORT_END=2023-11-14
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

### Issue

**Error during installation:**
```
[user@fedora /]$ sudo dkms build -m asus-wmi -v 1.0
Sign command: /lib/modules/6.0.7-301.fc37.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.0.7-301.fc37.x86_64 -C /lib/modules/6.0.7-301.fc37.x86_64/build M=/var/lib/dkms/asus-wmi/1.0/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.0.7-301.fc37.x86_64 (x86_64)
Consult /var/lib/dkms/asus-wmi/1.0/build/make.log for more information.
```

**make.log:**
``` 
DKMS make.log for asus-wmi-1.0 for kernel 6.0.7-301.fc37.x86_64 (x86_64)
Tue Mar 14 08:32:58 AM CET 2023
make: Entering directory '/usr/src/kernels/6.0.7-301.fc37.x86_64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.2.1 20220819 (Red Hat 12.2.1-2)
  You are using:           gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o
  CC [M]  /var/lib/dkms/asus-wmi/1.0/build/asus-nb-wmi.o
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c: In function ‘asus_wmi_input_init’:
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:538:61: error: ‘ASUS_WMI_DEVID_LID_FLIP_ROG’ undeclared (first use in this function); did you mean ‘ASUS_WMI_DEVID_LID_FLIP’?
  538 |                 result = asus_wmi_get_devstate_simple(asus, ASUS_WMI_DEVID_LID_FLIP_ROG);
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                             ASUS_WMI_DEVID_LID_FLIP
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:538:61: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c: In function ‘lid_flip_rog_tablet_mode_get_state’:
/var/lib/dkms/asus-wmi/1.0/build/asus-wmi.c:588:53: error: ‘ASUS_WMI_DEVID_LID_FLIP_ROG’ undeclared (first use in this function); did you mean ‘ASUS_WMI_DEVID_LID_FLIP’?
  588 |         result = asus_wmi_get_devstate_simple(asus, ASUS_WMI_DEVID_LID_FLIP_ROG);
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                     ASUS_WMI_DEVID_LID_FLIP
make[1]: *** [scripts/Makefile.build:249: /var/lib/dkms/asus-wmi/1.0/build/asus-wmi.o] Error 1
make: *** [Makefile:1856: /var/lib/dkms/asus-wmi/1.0/build] Error 2
make: Leaving directory '/usr/src/kernels/6.0.7-301.fc37.x86_64'
```

MechEng70, 2024-10-20:
Ran into the same issue on 6.11.3-200.fc40.x86_64

mjollnir14, 2025-01-28:
Same here

`pamac install asus-wmi-screenpad-dkms-git                                                         ✔  53s   TH  
Warning: asus-wmi-screenpad-dkms-git is only available from AUR
Preparing...
Cloning asus-wmi-screenpad-dkms-git build files...
Generating asus-wmi-screenpad-dkms-git information...
Checking asus-wmi-screenpad-dkms-git dependencies...
Resolving dependencies...
Checking inter-conflicts...

To build (1):
  asus-wmi-screenpad-dkms-git  r46.396c4b7-1    AUR

Edit build files : [e] 
Apply transaction ? [e/y/N] y

==== AUTHENTICATING FOR org.manjaro.pamac.commit ====
Authentication is required to install, update, or remove packages
Authenticating as: Ben (ben)
Password: 
==== AUTHENTICATION COMPLETE ====

Building asus-wmi-screenpad-dkms-git...
==> Making package: asus-wmi-screenpad-dkms-git r46.396c4b7-1 (mar. 28 janv. 2025 02:36:29)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating asus-wmi-screenpad-dkms-git git repo...
==> Validating source files with sha512sums...
    asus-wmi-screenpad-dkms-git ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of asus-wmi-screenpad-dkms-git git repo...
Cloning into 'asus-wmi-screenpad-dkms-git'...
done.
==> Starting prepare()...
Using: patch6.11 - version: 6.12
--2025-01-28 02:36:32--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.c?h=linux-6.12.y
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 145.40.73.55, 2604:1380:40e1:4800::1
Connecting to git.kernel.org (git.kernel.org)|145.40.73.55|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 134036 (131K) [text/plain]
Saving to: ‘asus-wmi.c’

asus-wmi.c            100%[=========================>] 130,89K   416KB/s    in 0,3s    

2025-01-28 02:36:32 (416 KB/s) - ‘asus-wmi.c’ saved [134036/134036]

--2025-01-28 02:36:32--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-wmi.h?h=linux-6.12.y
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 145.40.73.55, 2604:1380:40e1:4800::1
Connecting to git.kernel.org (git.kernel.org)|145.40.73.55|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2460 (2,4K) [text/plain]
Saving to: ‘asus-wmi.h’

asus-wmi.h            100%[=========================>]   2,40K  --.-KB/s    in 0s      

2025-01-28 02:36:33 (34,0 MB/s) - ‘asus-wmi.h’ saved [2460/2460]

--2025-01-28 02:36:33--  https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/plain/drivers/platform/x86/asus-nb-wmi.c?h=linux-6.12.y
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving git.kernel.org (git.kernel.org)... 145.40.73.55, 2604:1380:40e1:4800::1
Connecting to git.kernel.org (git.kernel.org)|145.40.73.55|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 19318 (19K) [text/plain]
Saving to: ‘asus-nb-wmi.c’

[... 133 lines cut ...]
  -> Running build hook: [kms]
==> WARNING: Possibly missing firmware for module: 'ast'
  -> Running build hook: [modconf]
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'wd719x'
==> WARNING: Possibly missing firmware for module: 'qla2xxx'
==> WARNING: Possibly missing firmware for module: 'qed'
==> WARNING: Possibly missing firmware for module: 'qla1280'
==> WARNING: Possibly missing firmware for module: 'bfa'
==> WARNING: Possibly missing firmware for module: 'aic94xx'
==> WARNING: Possibly missing firmware for module: 'xhci_pci'
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [encrypt]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating zstd-compressed initcpio image: '/boot/initramfs-6.6-x86_64-fallback.img'
  -> Early uncompressed CPIO image generation successful
==> Initcpio image generation successful
Log Orphan Packages                                                                                                                               [4/5]
Signal an update to the waybar pacman module after each update                                                                                    [5/5]
Transaction successfully finished.`

MechEng70, 2025-01-28:
Fixed my issue by uninstalling the older kernel and kernel-devel, rebooting and then reinstalling.