Plippo/asus-wmi-screenpad #52: Setting 0 to the brightness disconnect the screenpad instead of turning it black

sanjibukai, 2023-03-16:
Hello,
When I set the screenpad brightness to 0:
```
echo 0 | sudo tee '/sys/class/leds/asus::screenpad/brightness'
```
The screen became disconnected.
What I really want is to make the screen black while being here..
A workaround is to actually set it to 1..

Is this the correct behavior?
Is it possible to make it work like I want?

It seems on windows the dedicated key is doing that as well (well actually cycling like this: turning off the backlight -> disconnecting the screen -> connecting the screen again)

Thank you.

PTFOPlayer, 2023-03-16:
It is correct behavior, why do you want it to behave different?

sanjibukai, 2023-03-16:
I mean, I thought that would have been the default behavior (like it's the case in windows when the screen goes off)
The rationale is to switch off the screen but without disconnecting it so that opened applications and windows remain in place..
It's like when your screen goes into screensaver... This is just the screen that turns black (the backlight), but it's still connected, right?
One use case for example is when I want to watch a video on the main screen..
Currently setting the backlight to 1 is somehow doing the job even if when there is white content on the screenpad I can see it..
Anyway thank you, I guess I'll close the issue..

sanjibukai, 2023-03-16:
@PTFOPlayer Just saw your project `screenpad_rs`, is it working right now?