Plippo/asus-wmi-screenpad #62: did nothing except ruining boot

Ponali, 2024-06-30:
I'm currently using an ASUS VivoBook, with a Screenpad, dualbooting Linux Mint and Windows 11.
I wanted to have the features from the installed ASUS software from my windows partition to my linux partition, using the screenpad-tools repo, and started with this.
The first 4 steps were easy to get along, but then came the fifth. Apparently, i have to put in a password, different to Linux, so that third-party drivers could be used along with Secure boot.
I was a little scared, but i put in a new password anyway, and rebooted my computer.
I was greeted with a new boot menu, possibly regarding the Secure Boot shenanigans, and selected "Continue boot".
I finally got GRUB, but the resolution was scaled down to the minimum resolution (due to how my UEFI works, it scales it back to the whole monitor.)
I selected Linux Mint, waited a bit, and the resolution went back to normal.
I tried running the command from step 6, with "XXX" replaced to "0", but instead of having the brightness change, I got an error, saying the file doesn't exist.

[image]
[image]

Ponali, 2024-06-30:
i rebooted my computer again, and the Secure Boot menu dissapeared, and everything went like if nothing happened??

RobertoMaurizzi, 2024-07-08:
You need to add again your private Machine Operator Key with a command l;ike `sudo mokutil --import /var/lib/dkms/mok.pub` (replace the mok.pub with where you put your public key file) then you can check that the key has been scheduled for import using `sudo mokutil --list-new`, that should show the key.

At this point reboot and when the computer shows you the Secure Boot screen:
[image]
choose to enroll a new machine owner key and select your key form the list. Type the password and then the kernel should be able to load the module.

Another problem I had is that this module has the same name of the officlal one for Asus laptops, so I had to install it using `dkms install -m asus-wmi -v 1.0 --force`